bulletphysics/bullet3 #4388: How to properly generate .vtk file from .obj file?

ProgrammingElite, 2023-01-03:
I want to load deformable object in PyBullet with function "loadSoftBody". I know that PyBullet offers example object file "torus.obj" and "torus.vtk". It works well. However, when I try to generate .vtk file by myself from .obj with the following steps, PyBullet crashes down:
1. open .obj file in Blender in Windows and export .stl file
2. generate .msh file in TetWild (https://github.com/Yixin-Hu/TetWild) in Ubuntu with command in terminal:
$ ./TetWild --input torus.stl --output torus.msh
3. open .msh file in Gmsh in Windows and export .vtk file
4. run the following Python script:
```
import pybullet as p
from time import sleep
import pybullet_data

physicsClient = p.connect(p.GUI)

p.setAdditionalSearchPath(pybullet_data.getDataPath())

p.resetSimulation(p.RESET_USE_DEFORMABLE_WORLD)
p.setGravity(0, 0, -10)

planeId = p.loadURDF("plane.urdf", [0,0,0])

# original .vtk file ------- success
# softBodyId1 = p.loadSoftBody("original files/torus.vtk", basePosition = [0,0,5])
# generated .vtk file ------- fail
softBodyId2 = p.loadSoftBody("generated files/torus.vtk", basePosition = [0,0,5])

p.setRealTimeSimulation(0)

while p.isConnected():
p.stepSimulation()
```

Then PyBullet crashes down.
```
pybullet build time: May 20 2022 19:44:17
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=VMware, Inc.
GL_RENDERER=SVGA3D; build: RELEASE; LLVM;
GL_VERSION=4.1 (Core Profile) Mesa 21.2.6
GL_SHADING_LANGUAGE_VERSION=4.10
pthread_getconcurrency()=0
Version = 4.1 (Core Profile) Mesa 21.2.6
Vendor = VMware, Inc.
Renderer = SVGA3D; build: RELEASE; LLVM;
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0
MotionThreadFunc thread started
ven = VMware, Inc.
ven = VMware, Inc.
Segmentation fault (core dumped)
```

All the files are attached in the [files.zip](https://pybullet.org/Bullet/phpBB3/download/file.php?id=1966) 
Folder "original files" are files provided by PyBullet git repository. Folder "generated files" are files generated by myself. I search for the solution online and issue #2726 but **still cannot solve this problem**.

ProgrammingElite, 2023-01-03:
@erwincoumans

renatyv, 2023-01-08:
> All the files are attached in the [files.zip]
I generated .vtk file from your's and it works. Don't forget to put check the "save all elements" mark and choose "ASCII" format when exporting .vtk file using GMSH.

ProgrammingElite, 2023-01-08:
您的邮件已收到，谢谢！

ProgrammingElite, 2023-01-10:
I try it on Windows PyBullet and it works! But when I try it on VMware Ubuntu 20.04 PyBullet, the problem occurs. Why?

lidaken, 2023-04-01:
The process I translate to vtk is same to you, and I got error in Ubuntu PyBullet: malloc():memory corruption. Do you get the reason now?

ProgrammingElite, 2023-04-01:
您的邮件已收到，谢谢！

lucky3362, 2023-05-12:
please tell me how to fix this problem?

ProgrammingElite, 2023-05-12:
您的邮件已收到，谢谢！

lucky3362, 2023-05-15:
I have found the answer. When you use tetwild on Linux and Gmsh on Windows, this will result in an error. Only when both are used on Linux can success.