bulletphysics/bullet3 #4392: The deformable_torus example appears to be broken due to a vtk/urdf import error

danielpmorton, 2023-01-06:
Running `import pybullet_examples.deformable_torus` leads to starting a pybullet GUI window with a torus, and then a crash with the following message:
```
b3Printf: b3Warning[examples/SharedMemory/../Importers/ImportURDFDemo/UrdfFindMeshFile.h,102]:

b3Printf: : cannot find 'torus.vtk' in any directory in urdf path

Nodes:  0
Links:  0
Faces:  0
Tetras: 0
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/UrdfFindMeshFile.h,102]:

b3Printf: /usr/local/lib/python3.8/dist-packages/pybullet_data/torus_deform.urdf:12: cannot find 'torus.vtk' in any directory in urdf path

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/pybullet_examples/deformable_torus.py", line 22, in <module>
    bunny2 = p.loadURDF("torus_deform.urdf", [0,1,0.2], flags=p.URDF_USE_SELF_COLLISION)
pybullet.error: Cannot load URDF file.

```

Note that commenting out the lines in the source file associated with `torus.vtk`, `torus_deform.urdf`, and `bunny2` eliminates the crashing issue but this likely changes the desired behavior of the example

erwincoumans, 2023-02-24:
You need to clone the full bullet github repo and run the example from within that repo directory.

closed 2023-02-24 by erwincoumans

jeremitu, 2024-02-04:
Just find `torus.vtk` in `bullet3/data` and copy it into `Lib/site-packages/pybullet_data` or wherever `torus_deform.urdf` is.