bulletphysics/bullet3 #4396: Segfault in p.calculateJacobian()

DanSJohnson, 2023-01-18:
Hi everyone, 

I'm having issues where my calls to `p.getJacobian()` result in a segmentation fault. The code to recreate this is:
```
import pybullet as p
import numpy as np

phyID=p.connect(p.DIRECT)

robot_path='/basics/models/robot.urdf'
robotId = p.loadURDF(robot_path, flags=p.URDF_USE_SELF_COLLISION, physicsClientId=phyID)
p.stepSimulation()

jacobian=p.calculateJacobian(robotId, 7, [0, 0, 0], np.zeros(18) ,np.zeros(18),np.zeros(18), physicsClientId=phyID)
```

This results in the terminal output:
```
pybullet build time: May 20 2022 19:41:54
Segmentation fault (core dumped)
```
with no further information.

I am using pybullet version 3.17, python version 3.6.9 and my operating system is Ubuntu 18.04.5 LTS.

Any assistance would be greatly appreciated!

DanSJohnson, 2023-01-18:
Ok, I think i've found the source of the problem, it seems that using numpy arrays has somehow caused the seg fault?

When i replace the line:
`jacobian=p.calculateJacobian(robotId, 7, [0, 0, 0], np.zeros(18) ,np.zeros(18),np.zeros(18), physicsClientId=phyID)`
with:
`jacobian =p.calculateJacobian(robotId, 7, [0, 0, 0], np.zeros(18).tolist(), np.zeros(18).tolist(), np.zeros(18).tolist(), physicsClientId=phyID)`
I dont get a fault, and I am assuming that the result is indeed the correct jacobian.

closed 2023-01-18 by DanSJohnson

varadVaidya, 2023-03-03:
According to my experience, sending numpy arrays to pybullet functions causes some weird errors which are hard to trace.