bulletphysics/bullet3 #4400: Create half cylinder in PyBullet

Hs-Cheng, 2023-01-25:
Hi,

I would like to ask if it is possible to use the PyBullet function `createCollisionShape` to create a half cylinder in PyBullet? If so, how should I do that?

Or, is there any other function in PyBullet that can help me create a half cylinder?

The half cylinder is not a static object in the simulation space. It need to move (it should be a dynamic object and capable of collide with other objects).

Thank you very much for the help!

ManuCorrea, 2023-01-28:
Hello,

No it is not possible. You should create an STL/obj file with any 3D editor, this is a beginner friendly CAD editor that you could use -> https://www.tinkercad.com/

Then with the STL/obj you can so an URDF like int https://github.com/bulletphysics/bullet3/blob/master/data/cube.urdf

daniilpetrov, 2023-01-30:
And when I load models that are not primitives, I create either btGImpactMeshShape or btGImpactConvexDecompositionShape polygons. However, btGImpactConvexDecompositionShape can only be created for whole objects that do not contain holes because they cannot be passed through.

Hs-Cheng, 2023-02-14:
Thank you @ManuCorrea  and @daniilpetrov for your advices!

closed 2023-02-14 by Hs-Cheng