bulletphysics/bullet3 #4405: Drift in COM of object without any external force in bullet

pankhurivanjani, 2023-02-09:
Hello! 

I am using Pybullet to simulate the behavior of a human-like kinematic character with correct physics in a zero-gravity environment in  pybullet. 

I have Body articulations that I want to give to the kinematic character such that the motion is physically plausible. I don't have the root translation motion and expecting pybullet to simulate it. 

I am using bullet's setJointMotorControlArray function to give the joint angles to the kinematic character. Since this is based on PD controller, I am using it with a position control mode. There is a parameter in this function  "max_force". 
When this max_force value is 0, the kinematic character doesn't do any articulations, when I give some max_force value then it starts performing some motions. 

Now, my concern here is, with this max_force (which I am assuming is some internal force here), I observe the drift in the Center of Mass AND root of the character too other than the given articulations.  If we look into the physics, without any external force, the angular momentum  of COM should be conserved and there shouldn't be any drift in COM's position but if we are not applying any external force here, and if max_force value here just corresponds to the internal force required to reach the joint angle position then why am I observing the translation here?

Thanks,

erwincoumans, 2023-02-24:
Thanks for the report, but this is impossible to answer without a reproduction case to look at. Please re-open if you have some small pybullet python example + urdf to check.

closed 2023-02-24 by erwincoumans

pankhurivanjani, 2023-03-16:
Hello! 

This is a sample python script where I am using PD controller to pass MoCAP data in zero gravity conditions. 

```
def run_simulation(trans_1, clean_dofs1, dofs_1, model1):
    """
    Running the simulation for the model using the data obtained from the studio motion capture
    """
    
    for i in range(clean_dofs1.shape[0]):
        for k in range(100):
            p.setJointMotorControlArray(model1, jnt_ids, p.POSITION_CONTROL, targetPositions=clean_dofs1[i], targetVelocities = vel,forces=np.ones(72)*p.readUserDebugParameter(max_force), positionGains=np.ones(72)* p.readUserDebugParameter(kp), velocityGains=np.ones(72)* p.readUserDebugParameter(kd)) 

if __name__ == "__main__":   
    id_sim = p.connect(p.GUI)

    p.configureDebugVisualizer(p.COV_ENABLE_Y_AXIS_UP, 1)
    p.configureDebugVisualizer(flag=p.COV_ENABLE_MOUSE_PICKING, enable=1)
    p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 1)
    p.setGravity(0, 0, 0)
    p.setAdditionalSearchPath(pybullet_data.getDataPath())
    dofs_1, trans_1 = get_data(3, 0)
    model_path1 = "studio_new.urdf"
    startPos = trans_1[0] 
    model1 = p.loadURDF(model_path1, startPos, [1,0,0,1], useFixedBase=False,flags=p.URDF_USE_INERTIA_FROM_FILE)  #,flags=p.URDF_USE_INERTIA_FROM_FILE
    #model2 = p.loadURDF(model_path1, startPos, [1.5,0,0,1], useFixedBase=False,flags=p.URDF_USE_INERTIA_FROM_FILE)  #,flags=p.URDF_USE_INERTIA_FROM_FILE
    clean_dofs1, jnt_ids = data_arrangement(dofs_1)

run_simulation(trans_1, clean_dofs1, dofs_1, model1)
 
```