bulletphysics/bullet3 #4406: Contact Forces between rigid and soft bodies in PyBullet

HirakBasumatary, 2023-02-09:
I want to determine the contact forces between the hand and the soft body. I noticed that pybullet.getContactPoints(...) only gives me contact points between rigid bodies. 

The normal force result of pybullet.getContactPoints(...)  just gives "1", as the output in the case of contact between rigid and soft body. However, it runs fine when rigid body contacts rigid body. 

Has anyone gone through this problem in Pybullet ? I would be very grateful if the Pybullet community could guide me regarding this. 

`#!/usr/bin/env python3

import os
from matplotlib import scale
import numpy as np
import pybullet as p
import time
import pybullet_data
physicsClient = p.connect(p.GUI)#or p.DIRECT for non-graphical version
p.resetSimulation(p.RESET_USE_DEFORMABLE_WORLD)
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-1)
planeId = p.loadURDF("plane.urdf")
startPos = [0,0,-0.25]
startOrientation = [0,0,0.707,0.707]

tableId = p.loadURDF("urdf/cube.urdf",[0.1,0.65,2.5],[0,0,0,1],globalScaling=10)

for i in range(1000):
    p.stepSimulation()
    time.sleep(1./250.)

#Hirak Trial
softId=p.loadSoftBody("urdf/torus.vtk",[0.07,2.3,5.5],[0,0,1,1] , mass = 10 ,scale= 0.75 , useMassSpring=True, springElasticStiffness= 100, springDampingStiffness=100,springDampingAllDirections=100 )

for i in range(100):
    p.stepSimulation()
    time.sleep(1./250.)
handId = p.loadURDF("urdf/hand_T4.xacro",startPos, startOrientation,globalScaling=10)

ff=241
for i in range(10):
    p.stepSimulation()
    time.sleep(1./250.)
for i in range (1000):
    
    p.stepSimulation()
    p.setJointMotorControl2(handId,4,p.TORQUE_CONTROL,force = ff)
    p.setJointMotorControl2(handId,7,p.TORQUE_CONTROL,force = ff)
    p.setJointMotorControl2(handId,5,p.TORQUE_CONTROL,force = ff)
    p.setJointMotorControl2(handId,8,p.TORQUE_CONTROL,force = ff)
    
    time.sleep(1./250.)

for i in range (10000):
      
    p.stepSimulation()
    p.setJointMotorControl2(handId,2,p.POSITION_CONTROL,targetPosition=2,force = 50,velocityGain=100,positionGain=1)
    
    p.setJointMotorControl2(handId,4,p.TORQUE_CONTROL,force = ff)
    p.setJointMotorControl2(handId,7,p.TORQUE_CONTROL,force = ff)
    p.setJointMotorControl2(handId,5,p.TORQUE_CONTROL,force = ff)
    p.setJointMotorControl2(handId,8,p.TORQUE_CONTROL,force = ff)

    contactPoints=p.getContactPoints(softId, handId)
    for con in contactPoints:
        #nor+=con[9]
        print("messi")
        print(con[9])

    time.sleep(1./250.)

p.disconnect()

`

HirakBasumatary, 2023-02-10:
@erwincoumans @AlexanderFabisch @johbrust

AlexanderFabisch, 2023-02-10:
Unfortunately, some internal tests failed and the contact force calculation was removed from PyBullet.

We created a fork in which it is possible to obtain contact forces to soft bodies (and the full state of a tetrahedral mesh):

https://github.com/aprilprojecteu/bullet3

with commits of @johbrust and @MaartenBehn

johbrust, 2023-02-13:
In PyBullet [version 3.21](https://github.com/bulletphysics/bullet3/tree/3.21), the soft contact info was implemented. It got removed again for version 3.22 due to some internal tests. :man_shrugging:

I [commented on that issue](https://github.com/bulletphysics/bullet3/pull/3979), but never received an answer.

erwincoumans, 2023-02-24:
Let's try to re-enable, here is the pull request:
https://github.com/bulletphysics/bullet3/pull/4413

erwincoumans, 2023-02-24:
It has been merged and will be available in next pybullet release. Thanks!

closed 2023-02-24 by erwincoumans