bulletphysics/bullet3 #4418: project on macOS m2 vs Ubuntu on AMD and Nvidia

Roboticyang, 2023-02-25:
Hi, I am a researcher using pyBullet to work on some virtual works. I have both used ubuntu 22.04 and macOS enviroment (both python 3.10 and python 3.8) for the same setup. I noticed the following issues:

1. on my ubuntu laptop with opengl3, the GUI shows with broken UI (like in a diagonal half missing) and really small fonts. Opengl2 shows correct GUI but lost transparency on plane and UI fonts are still tiny. Although the above issue exits, the overall rendering is smooth. My customized physical engine uses constant computation time.
2. However, on my macOS m2 max laptop, the GUI rendering utilized almost 96+% of the GPU on opengl3, the GUI rendering and physical engine are really sluggish, although it shows metal support on the output.
3. My customized physical engine calculate force & torque between 10 objects. It costs around 5ms on my ubuntu laptop. But the time cost on mac variants by a large margin.

I am curious if this mac slowness issue is somewhat related to my settings. I used the quickstart guide to setup the connections with a few slide bars to control the control poses and some debug texts to show numbers. Everything else is my algorithm coding not related to the virtual environment. Please let me know if I need to provide more details. Thanks!

erwincoumans, 2023-04-26:
It is likely the shadow or transparency in rendering, or the text rendering. Try pressing 'g' or 's' and see if that helps.

closed 2023-04-26 by erwincoumans

Roboticyang, 2023-05-06:
@erwincoumans  Thanks alot, the g button to hide all utility columns worked for me. But I met another issue with the debug slider tool. On ubuntu it works fine. However, on macOS, there are quite frequent pybullet error message failed to load parameters of the debug sliders I created. I have around 5 of them, each time random slider shows the error message like below. Should I open another issue?
[image]