bulletphysics/bullet3 #4419: Weird idea: collisionshape subtraction.

Zincwing, 2023-02-28:
Hi,

I was thinking about collisionshapes, and I wondered if it would be possible to make a hole in an existing collisionshape by "subtracting" another shape. So if I have a wall with a round hole inside, I subtract a cilinder from the boxcollsionshape of the wall. Similar to how a CompoundCollisionShape is the sum of multiple shapes, the SubtractedCollisionShape or DifferenceCollisionshape would contain all points in one collisionshape that are not in the other.

The alternative would be a mesh collision or a weird compound collision containing all parts of the wall that are not in the hole. I think it would also simplify real time adding of holes to arbitrary shapes.

erwincoumans, 2023-04-26:
One option is to use a library for boolean operations, then generate triangle meshes from the surface and use those to perform collision detection. It can be done, but requires effort.

closed 2023-04-26 by erwincoumans

Zincwing, 2023-06-07:
Wouldn't it be possible to do something like this (pseudocode):

`
class CollsionShapeWithHole {
    mainCollisionShape: CollisionShape;
    hole: CollisionShape;
}

bool collidesWith(Collidible other) {
    return MainShape.collidesWith(other) && (!holeShape.CollidesWith(other))
}

`

With similar changes to narrow sweep tests and such?

BluePrintRandom, 2023-06-07:
there is a lib - bpy which imports headless blender,

you can use this for a bool operation I bet and get the evaluated mesh back triangulated and use it.