bulletphysics/bullet3 #4422: obj and vtk

lidaken, 2023-03-05:
Today I run deformable_ball.py, and I found that the ball will be broken(or be rigid?) when I delete simFileName = "ball.vtk" from the loadSoftBody.
So my question is that vtk file is the key to make ball deformable? Why it can do this and how to make vtk file for my obj file to make my own deformable model?
Thank you

rodriguesvinay, 2023-03-20:
Hey, as far as I understand .obj file is only for surface details (texture)--->useful for simulation of surface objects like cloth. For volumetric consideration, you need .vtk file with tetrahedral meshes. Being said that for volumetric simulation you don't need .obj file. you can create .obj file from most of the CAD software and for .vtk file please check [#2726](https://github.com/bulletphysics/bullet3/issues/2726)

lidaken, 2023-03-31:
> Hey, as far as I understand .obj file is only for surface details (texture)--->useful for simulation of surface objects like cloth. For volumetric consideration, you need .vtk file with tetrahedral [...]

Thanks a lot, sorry I reply your anwser too late. You are right, obj is surface and vtk is points inside. 
And now I get vtk file from obj to msh to vtk. But when I test my file in deformable_ball.py, I got: malloc():memory corruption. And I can't find the error in my file. Could you please read it or test it, thanks a lot again and again!
[givemetrue.zip](https://github.com/bulletphysics/bullet3/files/11123344/givemetrue.zip)

lidaken, 2023-04-13:
> Hey, as far as I understand .obj file is only for surface details (texture)--->useful for simulation of surface objects like cloth. For volumetric consideration, you need .vtk file with tetrahedral [...]

Thanks a lot, and I have already solved it!

lucky3362, 2023-05-12:
please tell me how to fix this problem?