bulletphysics/bullet3 #4428: How to set contact stiffness

lzq156525, 2023-03-14:
Hello,

Hello, I want to ask what is the principle of setting contact stiffness and contact damping in the changeDynamics() function, and are there any recommended parameters?

Best wishes,
Lan

erwincoumans, 2023-03-18:
See the Quickstart Guide: https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#
Contact stiffness and damping are converted into the LCP parameters constraint force mixing (CFM, a small value on the main diagonal of the LCP matrix A) and the Baumgate error reduction parameter (ERP) as explained in Erin Catto's Soft Constraints document: https://box2d.org/files/ErinCatto_SoftConstraints_GDC2011.pdf
Similar to a spring, those parameters interact with object mass, so it requires some tuning.

closed 2023-03-18 by erwincoumans

Akshay6077, 2024-06-20:
Hello,

I've encountered an unusual issue in my PyBullet simulation. Specifically, when setting the contactStiffness and contactDamping parameters in the changeDynamics() function, the behavior of the objects seems counterintuitive.
Description

    Expected Behavior: Higher values for contactStiffness and contactDamping should prevent objects from sinking into one another.
    Observed Behavior: When I set these parameters to -1, the box does not sink into the pedestal. However, when I set them to higher values, the box begins to sink, which is the opposite of what I expected.