bulletphysics/bullet3 #4439: malloc():memory corruption

lidaken, 2023-03-31:
Hello,
I use blender to create obj, then I use tetwild to build msh, and I use gmsh to create vtk file. But, when I test my vtk file in 
deformable_ball.py, It give me :malloc():memory corruption. So, can someone read my vtk file and point out something wrong? 

Thanks a lot!
[givemetrue.zip](https://github.com/bulletphysics/bullet3/files/11123229/givemetrue.zip)

danielpmorton, 2023-04-07:
I've gotten this issue before with a tetrahedral mesh I've made when I tried a different mesh generator other than GMSH. In general, the best workflow I've found is using GMSH on Ubuntu and manually deleting the lines corresponding to the triangular faces in the VTK file (and updating the CELLS and CELL_TYPES counts accordingly). 

You may also want to try a re-meshing of the object to improve regularity in the triangles. I use MeshMixer for this (on Windows) but I believe MeshLab on Linux will do something similar

lidaken, 2023-04-13:
> I've gotten this issue before with a tetrahedral mesh I've made when I tried a different mesh generator other than GMSH. In general, the best workflow I've found is using GMSH on Ubuntu and manually deleting the lines corresponding to the triangular faces in the VTK file (and updating the CELLS and CELL_TYPES counts accordingly).
> 
> You may also want to try a re-meshing of the object to improve regularity in the triangles. I use MeshMixer for this (on Windows) but I believe MeshLab on Linux will do something similar
> 
> Apparently there is a way to export only the tetrahedral elements in GMSH, but I didn't figure this out yet. Since I don't need to make a ton of meshes, it wasn't a high priority for me

Thanks a lot, and I have already deal the problem! Using tetwild and gmsh both in Ubuntu can sovle the problem quickly.

lucky3362, 2023-05-12:
please tell me how to fix this problem

lucky3362, 2023-05-15:
I have found the answer. When you use tetwild on Linux and Gmsh on Windows, this will result in an error. Only when both are used on Linux can success.