bulletphysics/bullet3 #4445: Unreliable velocity and angular velocity for softbody

danielpmorton, 2023-04-11:
Calling `getBaseVelocity` on a softbody does not seem to produce meaningful results in Pybullet. Interestingly, I tried printing this info via modifying the C++ source, and I was able to get velocity values out of the soft torus example in the ExampleBrowser. I tried modifying the source but could not figure out precisely how this is communicated from the C++ to Python

This issue also exists when I build pybullet from source rather than using pip

Here is a minimal example of printing the velocity information that just returns vec3's of zeros

```
import time
import pybullet
import pybullet_data

pybullet.connect(pybullet.GUI)
pybullet.resetSimulation(pybullet.RESET_USE_DEFORMABLE_WORLD)
pybullet.setGravity(0, 0, -9.81)
pybullet.setAdditionalSearchPath(pybullet_data.getDataPath())
sb = pybullet.loadSoftBody(
    "/home/dan/software/bullet3/data/bread.vtk",  # Could not use pybullet_data path for some reason
    mass=3,
    useNeoHookean=1,
    NeoHookeanMu=180,
    NeoHookeanLambda=600,
    NeoHookeanDamping=0.01,
    collisionMargin=0.006,
    useSelfCollision=1,
    frictionCoeff=0.5,
    repulsionStiffness=800,
)
pybullet.loadURDF("plane.urdf", basePosition=[0, 0, -2])
while True:
    pybullet.stepSimulation()
    time.sleep(1 / 120)
    vel, ang_vel = pybullet.getBaseVelocity(sb)
    print("Velocity: ", vel)
    print("Angular velocity: ", ang_vel)
```

erwincoumans, 2023-04-20:
Confirmed, many APIs were developed purely for multibodies. Once the softbody/deformables were added, those APIs haven't been implemented.

danielpmorton, 2023-05-08:
Hey Erwin, I decided to implement this - here's the PR https://github.com/bulletphysics/bullet3/pull/4465/
Let me know if there is anything you'd like changed, or if there's anything important I missed!