bulletphysics/bullet3 #4451: Incorrect joint force/torque sensor values for a passive pendulum

azeey, 2023-04-19:
Might be related to #2966, but no motor torque is applied here.

We've run into an issue with implementing force/torque sensors for Bullet in gz-physics https://github.com/gazebosim/gz-physics/pull/434 where test expectations that pass with DART are failing with Bullet (`btMultiBody`). To narrow down the scope, I've set up a simple example that demonstrates the problem. The example loads a URDF of simple pendulum that passively swings due to gravity starting from an initial angle. It prints the "sensed" force as well as the expected reaction forces. The following is the last printout of the script below showing that the normal (centripetal) forces sensed by bullet are significantly different than the expected value.

```
joint pos: -0.0005009165033698082, vel: -4.62124490737915, accel: -0.0438690185546875
 sensor force: (0.0, 0.004692971706390381, 289.4410705566406) norm: 289.4410705946863
 expected force: [0, -0.2663465830861952, 174.1218738355234] norm: 174.12207754473943
```

However, applying the following patch seems to fix the issue (credit goes to @nkoenig)
```diff
diff --git a/src/BulletDynamics/Featherstone/btMultiBody.cpp b/src/BulletDynamics/Featherstone/btMultiBody.cpp
index d7588aedc..fcadfbb65 100644
--- a/src/BulletDynamics/Featherstone/btMultiBody.cpp
+++ b/src/BulletDynamics/Featherstone/btMultiBody.cpp
@@ -936,7 +936,10 @@ void btMultiBody::computeAccelerationsArticulatedBodyAlgorithmMultiDof(btScalar
 			if (m_useGyroTerm)
 				zeroAccSpatFrc[i + 1].addAngular(spatVel[i + 1].getAngular().cross(m_links[i].m_inertiaLocal * spatVel[i + 1].getAngular()));
 			//
-			zeroAccSpatFrc[i + 1].addLinear(m_links[i].m_mass * spatVel[i + 1].getAngular().cross(spatVel[i + 1].getLinear()));
+			if (!isConstraintPass)
+			{
+				zeroAccSpatFrc[i + 1].addLinear(m_links[i].m_mass * spatVel[i + 1].getAngular().cross(spatVel[i + 1].getLinear()));
+			}
 			//
 			//btVector3 temp = m_links[i].m_mass * spatVel[i+1].getAngular().cross(spatVel[i+1].getLinear());
 			////clamp parent's omega
```

<details><summary>pendulum.urdf</summary>

```xml
<?xml version="1.0"?>
<robot name="pendulum">
  <link name="base_link">
    <visual>
      <geometry>
        <sphere  radius="0.1"/>
      </geometry>
      <material name="blue">
        <color rgba="0 0 .8 1"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <sphere  radius="0.1"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="10"/>
      <inertia ixx="1.0" ixy="0.0" ixz="0.0" iyy="1.0" iyz="0.0" izz="1.0"/>
    </inertial>
  </link>

  <joint name="pivot" type="continuous">
    <axis xyz="1 0 0"/>
    <parent link="base_link"/>
    <child link="bob"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <limit effort="0"/>
    <joint_properties damping="0.0" friction="0.0"/>
  </joint>

  <link name="bob">
    <visual>
      <origin xyz="0 0 -0.9"/>
      <geometry>
        <sphere radius="0.2"/>
      </geometry>
      <material name="white"/>
    </visual>
    <collision>
      <origin xyz="0 0 -0.9"/>
      <geometry>
        <sphere radius="0.2"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="6"/>
      <origin xyz="0 0 -0.9"/>
      <inertia ixx="0.04" ixy="0.0" ixz="0.0" iyy="0.04" iyz="0.0" izz="0.04"/>
    </inertial>
  </link>
</robot>
```
</details> 

<details><summary>test_pendulum_reaction.py</summary>

```python
import pybullet as p
import math
from math import cos, sin
from numpy.linalg import norm
from time import sleep

#  physicsClient = p.connect(p.GUI)
physicsClient = p.connect(p.DIRECT)

pendulumId = p.loadURDF("pendulum.urdf", useFixedBase=True)
jointIndex = p.getNumJoints(pendulumId) - 1
bobLinkIndex = p.getNumBodies(pendulumId) - 1
p.setJointMotorControl2(pendulumId, jointIndex, controlMode=p.VELOCITY_CONTROL, force=0)
p.resetJointState(pendulumId, jointIndex, math.pi / 2)
p.changeDynamics(pendulumId, bobLinkIndex, linearDamping=0, angularDamping=0)
p.enableJointForceTorqueSensor(pendulumId, jointIndex, enableSensor=1)

linkState = p.getLinkState(pendulumId, bobLinkIndex)
linkDynamics = p.getDynamicsInfo(pendulumId, bobLinkIndex)

r = norm(linkState[2]) # distance from joint to COM
m = linkDynamics[0] # mass of pendulum bob
g = 9.8 # magnitude of gravity
print(f"r {r} m {m}")

p.setGravity(0, 0, -g)
omega_prev = 0

while p.getJointState(pendulumId, jointIndex)[0] > 0:
  dt = 0.001
  p.setTimeStep(dt)
  p.stepSimulation()
  sleep(1.0/240)
  jointState = p.getJointState(pendulumId, jointIndex)
  force = jointState[2][:3]

  # Given the position (θ), velocity (ω), and acceleration (α) of the joint
  # and distance from the joint to the COM (r), the reaction forces in
  # the tangent direction (Ft) and normal direction (Fn) are given by:

  # Ft = m * α * r + (m * g * sin(θ))  = m * (α * r + g * sin(θ))
  # Fn = m * ω² * r + (m * g * cos(θ)) = m * (ω² * r + g * cos(θ))

  theta = jointState[0]
  omega = jointState[1]
  alpha = (omega - omega_prev)/dt
  force_t = m * (alpha * r + g * sin(theta))
  force_n = m * (omega**2 * r + g * cos(theta))

  # The orientation of the joint frame is such that the normal force is
  # parallel to the z-axis and the tangent force is parallel to the y-axis.
  expected_force = [0, force_t, force_n]
  print(f"joint pos: {theta}, vel: {omega}, accel: {alpha}\n"
        f" sensor force: {force} norm: {norm(force)}\n"
        f" expected force: {expected_force} norm: {norm(expected_force)}")

  omega_prev = omega
```
</details>

erwincoumans, 2023-04-20:
Thanks a lot for reporting this, tracking it down and proposing a fix!
Also, very nice to see that Gazebo gets the option for Bullet Featherstone articulations!

azeey, 2023-04-24:
> Thanks a lot for reporting this, tracking it down and proposing a fix! Also, very nice to see that Gazebo gets the option for Bullet Featherstone articulations!

You're welcome. And, yes, we're excited to be able to use Bullet-Featherstone. 

Can you confirm if the proposed fix a good solution? If so I'd be happy to create a PR.

erwincoumans, 2023-04-26:
I think the solution is acceptable. In a nutshell, in PyBullet we run ABA twice, the first time to get unconstrained accelerations. Then we run the constraint solver (iterative impulse/PGS by default). After that, we know the constraint forces, and run ABA to compute the joint forces taking the constraint forces into account. I haven't investigated properly, but it could be that we don't need to run the coriolis/spatial forces a second time.

Yes, please create a PR and I merge it.

nkoenig, 2023-04-26:
Thanks Erwin. PR i#4462 is open for your review.

closed 2023-05-04 by erwincoumans