bulletphysics/bullet3 #4485: [bug]: pybullet not installing in google colab

bheijden, 2023-06-23:
Google Colab hangs on installing pybullet with pip with the following output:

Running the following in google colab:
```
!pip3 install pybullet --upgrade
```
Results in the following output:
```
Looking in indexes: https://pypi.org/simple, https://us-python.pkg.dev/colab-wheels/public/simple/
Collecting pybullet
  Downloading pybullet-3.2.5.tar.gz (80.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 80.5/80.5 MB 12.1 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: pybullet

```

To reproduce, you can run the [colab](https://colab.research.google.com/drive/15JSROMJbeiqxcUwifPR2NYeeFBKmyIlX#scrollTo=E2eWDjPZsQc5) advertised on [this page](https://pybullet.org/wordpress/index.php/2021/04/15/pybullet-in-a-colab/).

jelledouwe, 2023-07-06:
It seems that installing pybullet on colab is very slow, because there is no wheel available on PyPI for python 3.10 (see https://pypi.org/project/pybullet/#files). Therefore pybullet is built from source which takes some time.

behradkhadem, 2023-08-06:
I had issues with this matter in the past weeks, turns out that the package installation takes 9 minutes. If you wait that long it'll work without a hassle.