bulletphysics/bullet3 #4509: Failed to create an OpenGL context

dkapur17, 2023-08-13:
I am trying to get PyBullet to work on my WSL2 installation of Ubuntu 22.04 with an Nvidia GTX 1050Ti. When I run `p.connect(p.GUI)`, the execution fails with the following output:

```
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
libGL error: MESA-LOADER: failed to open swrast: /home/dkapur17/anaconda3/envs/drl/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Creating context
Failed to create GL 3.3 context ... using old-style GLX context
Failed to create an OpenGL context
```

Following is the output of `glxinfo -B`:

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce GTX 1050 Ti) (0xffffffff)
    Version: 23.0.4
    Accelerated: yes
    Video memory: 12132MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 4.2
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce GTX 1050 Ti)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL shading language version string: 4.20
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.0.4-0ubuntu1~22.04.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
```

And the output of  `nvidia-smi`:

```
Sun Aug 13 17:29:16 2023
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.54.06              Driver Version: 536.40       CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1050 Ti     On  | 00000000:01:00.0  On |                  N/A |
| N/A   44C    P8              N/A / ERR! |    645MiB /  4096MiB |      6%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A        23      G   /Xwayland                                 N/A      |
+---------------------------------------------------------------------------------------+
```

Any help would be appreciated!

dkapur17, 2023-08-15:
Solved! For others facing the same issue in the future, it was a dependency issue, that I solved by installing the required stuff:

```
conda install -c conda-forge libgcc=5.2.0
conda install -c anaconda libstdcxx-ng
conda install -c conda-forge gcc=12.1.0
```

closed 2023-08-15 by dkapur17

Beyond0128, 2024-02-19:
thank you very much！！！

jayaramreddy10, 2024-05-07:
> Solved! For others facing the same issue in the future, it was a dependency issue, that I solved by installing the required stuff:
> 
> ```
> conda install -c conda-forge libgcc=5.2.0
> conda install -c anaconda libstdcxx-ng
> conda install -c conda-forge gcc=12.1.0
> ```

Thanks, it solved

fengjay01, 2025-04-12:
> Solved! For others facing the same issue in the future, it was a dependency issue, that I solved by installing the required stuff:解决了!对于将来面临同样问题的其他人来说，这是一个依赖问题，我通过安装所需的东西来解决：
> 
> ```
> conda install -c conda-forge libgcc=5.2.0
> conda install -c anaconda libstdcxx-ng
> conda install -c conda-forge gcc=12.1.0
> ```

thank you very much！！！