bulletphysics/bullet3 #4523: pip-installed PyBullet does not have NumPy enabled

luator, 2023-09-11:
According to the documentation "pip install pybullet has NumPy enabled, if available on the system."
However, I just ran into an issue that indicates that NumPy is not used (`getCameraImage` returned an tuple instead of the expected array).  Also `pybullet.isNumpyEnabled()` returns 0.

Both PyBullet and NumPy are installed via pip (in a venv).  I did not have this issue in the past, so could it be a problem that was introduced with a recent release (of either PyBullet or NumPy)?  Or am I missing something?

Steps to reproduce:
```bash
python3 -m venv /tmp/test-venv
. /tmp/test-venv/bin/activate
pip install -U pip

pip install numpy
pip install pybullet

python -c "import importlib.metadata; print('NumPy version:', importlib.metadata.version('numpy'))"
python -c "import importlib.metadata; print('PyBullet version:', importlib.metadata.version('pybullet'))"
python -c "import pybullet; print('NumPy enabled:', pybullet.isNumpyEnabled())"
```
Output I get (on Ubuntu 22.04 with Python 3.10.12):
```
...
NumPy version: 1.25.2
PyBullet version: 3.2.5
pybullet build time: Aug 24 2023 11:18:40
NumPy enabled: 0
```

luator, 2023-09-11:
I do not have the issue in a different environment (using a container) with an older NumPy version.  There I get
```
NumPy version: 1.23.3
PyBullet version: 3.2.5
pybullet build time: Aug 23 2023 11:35:33
NumPy enabled: 1
```
However, just downgrading NumPy in the "steps to reproduce" above does not help.  I noticed that even though the PyBullet version is the same, the build time differs by one day, so it seems that's actually also not exactly the same version.

luator, 2024-01-10:
I just noticed that this seems to be fixed in the meantime.  If I repeat the steps from above now, I get the following output (note that NumPy enabled is 1 now, as expected):
```
NumPy version: 1.26.3
PyBullet version: 3.2.6
pybullet build time: Nov 28 2023 23:45:17
NumPy enabled: 1
```

sqdy2669, 2024-07-17:
how can i get NumPy version: 1.26.3

luator, 2025-02-12:
Same thing happening again now with the latest release:
```
NumPy version: 2.2.2
PyBullet version: 3.2.7
pybullet build time: Feb 12 2025 10:10:02
NumPy enabled: 0
```

xiaofeifei-1, 2025-02-25:
Usefull

> According to the documentation "pip install pybullet has NumPy enabled, if available on the system." However, I just ran into an issue that indicates that NumPy is not used (`getCameraImage` returned an tuple instead of the expected array). Also `pybullet.isNumpyEnabled()` returns 0.
> 
> Both PyBullet and NumPy are installed via pip (in a venv). I did not have this issue in the past, so could it be a problem that was introduced with a recent release (of either PyBullet or NumPy)? Or am I missing something?
> 
> Steps to reproduce:
> 
> python3 -m venv /tmp/test-venv
> . /tmp/test-venv/bin/activate
> pip install -U pip
> 
> pip install numpy
> pip install pybullet
> 
> python -c "import importlib.metadata; print('NumPy version:', importlib.metadata.version('numpy'))"
> python -c "import importlib.metadata; print('PyBullet version:', importlib.metadata.version('pybullet'))"
> python -c "import pybullet; print('NumPy enabled:', pybullet.isNumpyEnabled())"
> Output I get (on Ubuntu 22.04 with Python 3.10.12):
> 
> ```
> ...
> NumPy version: 1.25.2
> PyBullet version: 3.2.5
> pybullet build time: Aug 24 2023 11:18:40
> NumPy enabled: 0
> ```

**Usefull**