bulletphysics/bullet3 #4538: The cable model exploded.

madejiaaha, 2023-11-02:
[image]
Hello everyone,

I'm a newcomer to PyBullet and I'm trying to simulate the action of a gripper grabbing a cable using PyBullet. To better represent the deformation of the cable, I imported the cable model as a SoftBody into my project. I'm currently facing an issue where the gripper does not interact correctly with the cable. I've tried adjusting the parameters of the cable, but the results I get after running the simulation fall into two categories:

Once the gripper comes into contact with the cable, the cable explodes (as shown in the image).
The gripper's model simply passes through the cable's model, and the cable doesn't deform in response.
Here are the current physical parameters of the cable (in the case where the cable explodes upon contact):

kabelId = p.loadSoftBody("M:/Hiwi/Pybullet/Modelle/Kabel/Tetild/Kabel_05_sf.obj", 
                         simFileName="M:/Hiwi/Pybullet/Modelle/Kabel/TetWild/Kabel_05.vtk", 
                         scale=0.4,
                         basePosition=[5, -15, -15],
                         mass=0.1,
                         collisionMargin=0.002,
                         useMassSpring=1,
                         useBendingSprings=1,
                         useNeoHookean=0,
                         springElasticStiffness=300,
                         springDampingStiffness=5,
                         springBendingStiffness=0.5,
                         frictionCoeff=1,
                         useFaceContact=0,
                         useSelfCollision=1,
                         repulsionStiffness=0.1)

Could anyone please tell me what might be causing this issue and how I can go about resolving it? Thank you!

closed 2023-11-06 by madejiaaha

madejiaaha, 2023-11-08:
The cable model exploded because the parameters of the cable were set incorrectly, and I solved the problem by setting the parameters as follows
kabelId = p.loadSoftBody("M:/Hiwi/Pybullet/Modelle/Kabel/Tetild/Kabel_05_sf.obj", 
                         simFileName="M:/Hiwi/Pybullet/Modelle/Kabel/TetWild/Kabel_05.vtk", 
                         scale = 0.4,
                         mass = 0.5,
                         basePosition = [5,-15,-15],
                         useMassSpring = 1,
                         useNeoHookean = 1, 
                         NeoHookeanMu = 100, 
                         NeoHookeanLambda = 300, 
                         NeoHookeanDamping = 0.4, 
                         collisionMargin = 0.002, 
                         useSelfCollision = 1, 
                         frictionCoeff = 1, 
                         repulsionStiffness = 800)

unishuai, 2024-07-27:
> The cable model exploded because the parameters of the cable were set incorrectly, and I solved the problem by setting the parameters as follows kabelId = [...]

hello,
I'm a newcomer to PyBullet,too.i meet the same question.Can i know how to deal this problem with you。thank you.