bulletphysics/bullet3 #4555: All initial letters of the text cannot be displayed correctly in Bullet GUI or PyBullet GUI.

Pengdage999, 2023-12-08:
As shown in the picture below, font display has encountered an error.
Is there any solutions?
[image]

cuppajoeman, 2023-12-18:
I can also confirm this, running on linux

salehrayan, 2024-05-22:
same here running on Windows. also the letters completely disappear if p.addUserDebugPoints is used but not if p.addUserDebugLines is used. Not sure if they are related.

justincredible, 2025-01-27:
This PR has been linked in the issue @cuppajoeman mentioned, but I thought I'd mention it here too if anyone was interested. I just noticed this, but I think @salehrayan's comment is relevant as it seems related to the change where this occurred.