bulletphysics/bullet3 #4587: Pybullet: more object needed

yqi19, 2024-03-01:
Hi, I would like to sincerely ask you a question,
in pybullet I require a table with white color and a shelf, a desk, and some other objects but they are missing.

May I ask how to acquire these?

eduardo98m, 2024-04-11:
You need to create the 3d models (.obj,.stl, etc) of the objects you want i  the simulation and add them via an .urdf file or by directly loading the meshes as collision/visual objects.

eduardo98m, 2024-04-11:
You can create the objects you want in a 3d modeling software like blender or CAD software like FreeCAD or Fusion360. There might be models available in public repositories like grab-CAD but you need to check the file format and the license of the model (In case you want to use the model for commercial purposes).