bulletphysics/bullet3 #4590: Build fails with strict-aliasing violations

eli-schwartz, 2024-03-10:
I tried to compile with LTO: `-flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

The -Werror=* flags are important to detect cases where the compiler can try to optimize based on assuming UB cannot happen, and miscompile code that has UB in it. strict-aliasing issues are always bad but LTO can make them even worse.

I got this error:

```
[1/16] Building CXX object src/BulletCollision/CMakeFiles/BulletCollision.dir/Gimpact/gim_memory.o
FAILED: src/BulletCollision/CMakeFiles/BulletCollision.dir/Gimpact/gim_memory.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBT_THREADSAFE=1 -DBT_USE_EGL -DBT_USE_OPENMP=1 -DBulletCollision_EXPORTS -DNO_OPENGL3 -I/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src  -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-clash-protection -flto=8 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -std=c++11 -fopenmp -fPIC -MD -MT src/BulletCollision/CMakeFiles/BulletCollision.dir/Gimpact/gim_memory.o -MF src/BulletCollision/CMakeFiles/BulletCollision.dir/Gimpact/gim_memory.o.d -o src/BulletCollision/CMakeFiles/BulletCollision.dir/Gimpact/gim_memory.o -c /var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_memory.cpp
In file included from /var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_memory.h:35,
                 from /var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_memory.cpp:30:
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h: In function ‘btScalar gim_inv_sqrt(btScalar)’:
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:137:22: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  137 |         GIM_INV_SQRT(f, r);
      |                      ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:74:29: note: in definition of macro ‘GIM_IR’
   74 | #define GIM_IR(x) ((GUINT&)(x))
      |                             ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:137:9: note: in expansion of macro ‘GIM_INV_SQRT’
  137 |         GIM_INV_SQRT(f, r);
      |         ^~~~~~~~~~~~
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:122:39: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  122 |                         isva = GIM_FR(_y);                         \
      |                                       ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:83:29: note: in definition of macro ‘GIM_FR’
   83 | #define GIM_FR(x) ((GREAL&)(x))
      |                             ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:137:9: note: in expansion of macro ‘GIM_INV_SQRT’
  137 |         GIM_INV_SQRT(f, r);
      |         ^~~~~~~~~~~~
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h: In function ‘btScalar gim_sqrt(btScalar)’:
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:144:18: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  144 |         GIM_SQRT(f, r);
      |                  ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:74:29: note: in definition of macro ‘GIM_IR’
   74 | #define GIM_IR(x) ((GUINT&)(x))
      |                             ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:129:17: note: in expansion of macro ‘GIM_INV_SQRT’
  129 |                 GIM_INV_SQRT(va, sva); \
      |                 ^~~~~~~~~~~~
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:144:9: note: in expansion of macro ‘GIM_SQRT’
  144 |         GIM_SQRT(f, r);
      |         ^~~~~~~~
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:122:39: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  122 |                         isva = GIM_FR(_y);                         \
      |                                       ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:83:29: note: in definition of macro ‘GIM_FR’
   83 | #define GIM_FR(x) ((GREAL&)(x))
      |                             ^
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:129:17: note: in expansion of macro ‘GIM_INV_SQRT’
  129 |                 GIM_INV_SQRT(va, sva); \
      |                 ^~~~~~~~~~~~
/var/tmp/portage/sci-physics/bullet-3.25/work/bullet3-3.25/src/BulletCollision/Gimpact/gim_math.h:144:9: note: in expansion of macro ‘GIM_SQRT’
  144 |         GIM_SQRT(f, r);
      |         ^~~~~~~~
cc1plus: some warnings being treated as errors
ninja: build stopped: cannot make progress due to previous errors.
```
and similar. Github won't let me post the entire error as it is too long...

Original downstream report (against 3.21, but as noted above I verified it on 3.25 as well): https://bugs.gentoo.org/863275
Full build log including all 8 miscompiling files: [build.log](https://github.com/bulletphysics/bullet3/files/14551735/build.log)

erwincoumans, 2024-04-09:
Thanks for the fixes. I've suspended work on Bullet/PyBullet for now, except for high-urgency fixes.
Since the Super Tux Kart project uses an old version, I would recommend making these changes over there (they seem to not use system Bullet). https://github.com/supertuxkart/stk-code/issues/5035

eli-schwartz, 2024-04-10:
Unfortunately fixing it just in supertuxkart doesn't help. I discovered the supertuxkart issue later, and mostly unrelatedly. Fixing supertuxkart won't help for other software that does link to (a single system copy of) bullet, such as godot, ogre, efl, openmw, gazebo...

Since someone else has offered a PR to implement the fixes, it seems like it should be relatively easy to review those fixes and merge them. The vast majority of development effort has been expended by someone else. :)