bulletphysics/bullet3 #4595: python 3.12 libraries are not found by cmake

ncopa, 2024-03-26:
When trying to build bullet 3.25 on Alpine Linux with python 3.12 I get the following warnings and errors:

```
...
OPENGL FOUND
/usr/lib/libGL.so/usr/lib/libGLU.so
-- Found PythonInterp: /usr/bin/python3 (found version "3.12.2")
-- Looking for versions: 3.12.2;3.12
-- Looking for python version '3.12.2' by checking executables: /usr/bin/python3;python;python3;python3.12.
CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3.12, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

-- Looking for python version '3.12' by checking executables: /usr/bin/python3;python;python3;python3.12.
CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3.12, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:349 (FIND_PACKAGE)

-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.12.2")
-- Looking for versions: 3.12;3.6;3.5;3.4;3.3;3.2;3.1;3.0;2.7;2.6;2.5;2.4;2.3;2.2;2.1;2.0;1.6;1.5
-- Looking for python version '3.12' by checking executables: /usr/bin/python3;python;python3;python3.12.
CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:357 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:357 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3, but could not extract version info.
Call Stack (most recent call first):
  CMakeLists.txt:357 (FIND_PACKAGE)

CMake Warning at build3/cmake/FindPythonLibs.cmake:211 (message):
  Found executable /usr/bin/python3.12, but could not extract version info.
[... 224 lines cut ...]

-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) 
-- Found NumPy: /usr/lib/python3.12/site-packages/numpy/core/include (found version "1.25.2")
-- NumPy ver. 1.25.2 found (include: /usr/lib/python3.12/site-packages/numpy/core/include)
NumPy found
-- Configuring done (1.2s)
-- Generating done (0.0s)
-- Build files have been written to: /home/ncopa/aports/community/bullet/src/bullet3-3.25/build
[1/902] Building CXX object src/Bullet3Serialize/Bullet2FileLoader/CMakeFiles/Bullet2FileLoader.dir/b3Serializer.o
...
[83/902] Building CXX object examples/TwoJoint/CMakeFiles/App_TwoJoint.dir/__/ThirdPartyLibs/clsocket/src/ActiveSocket.o
[84/902] Building CXX object examples/TwoJoint/CMakeFiles/App_TwoJoint.dir/__/ThirdPartyLibs/clsocket/src/SimpleSocket.o
ninja: job failed: /usr/bin/gcc -DBT_ENABLE_CLSOCKET -DBT_ENABLE_ENET -DBT_ENABLE_VHACD -DBT_THREADSAFE=1 -DBT_USE_DOUBLE_PRECISION -DBT_USE_EGL -DHAS_SOCKLEN_T -DPYBULLET_USE_NUMPY -DSTATIC_LINK_SPD_PLUGIN -DUSE_GRAPHICAL_BENCHMARK -D_LINUX -Dpybullet_EXPORTS -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/src -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/ThirdPartyLibs -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/ThirdPartyLibs/enet/include -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/ThirdPartyLibs/clsocket/src -I/usr/lib/python3.12/site-packages/numpy/core/include -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/pybullet/../../Extras/VHACD/inc -I/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/pybullet/../../Extras/VHACD/public -Os -fstack-clash-protection -Wformat -Werror=format-security -fno-plt -O2 -DNDEBUG -flto=auto -fPIC -MD -MT examples/pybullet/CMakeFiles/pybullet.dir/pybullet.o -MF examples/pybullet/CMakeFiles/pybullet.dir/pybullet.o.d -o examples/pybullet/CMakeFiles/pybullet.dir/pybullet.o -c /home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/pybullet/pybullet.c
/home/ncopa/aports/community/bullet/src/bullet3-3.25/examples/pybullet/pybullet.c:40:10: fatal error: Python.h: No such file or directory
   40 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.
ninja: subcommand failed
```

Please note that python3 is there and can easily be found by `pkg-config`:
```
$ pkg-config --cflags python3
-I/usr/include/python3.12
```

ncopa, 2024-03-26:
Problem is that python 3.12 no longer bundles `distutils` which is used by `build3/cmake/FindLibPython.py`.

jakirkham, 2025-02-10:
`setuptools` contains a vendored copy of `distutils`