bulletphysics/bullet3 #4649: Numpy 2.x Importing Error

hilookas, 2024-08-02:
```log
pybullet build time: Nov 28 2023 23:45:17

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.0.1 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "/home/rosdev/ros2_ws/non_ros_src/mr_urdf_loader/tests/example/3DoF/sim.py", line 1, in <module>
    import pybullet as p
AttributeError: _ARRAY_API not found
Traceback (most recent call last):
  File "/home/rosdev/ros2_ws/non_ros_src/mr_urdf_loader/tests/example/3DoF/sim.py", line 1, in <module>
    import pybullet as p
ImportError: numpy.core.multiarray failed to import
```

Minimal reproducible example:

```
pip install -U numpy
pip install -U pybullet
```

```python
import numpy
import pybullet
```

araffin, 2024-11-19:
@erwincoumans I could build pybullet with numpy 2.0 locally without any problem (but I didn't find the many linux wheel builder script to share a pre-built wheel with others)

For the others, if you want to use pybullet with numpy 2, you can do `pip install git+https://github.com/bulletphysics/bullet3` to build from source.

Marc-Morcos, 2024-11-19:
@erwincoumans so I guess its just a matter of building a new wheel for pypi?

eaa3, 2025-01-13:
> @erwincoumans so I guess its just a matter of building a new wheel for pypi?

Hi guys, just wondering do we have any updates about this?

erwincoumans, 2025-01-30:
I haven't gotting back to Bullet for a while. Just uploaded new pip packages ([pybullet-3.2.7](https://pypi.org/project/pybullet/)) compiled against numpy 2.x (should work with numpy 1.x as well).

Thanks!
Erwin

closed 2025-01-30 by erwincoumans