bulletphysics/bullet3 #4650: removebody cannot remove createMultibody

unishuai, 2024-08-07:
i use createMultiBody to create a cable,And i need to remove it.when i  use removebody to deal it,it unfortunately do not work.my simple  code is there.
`# ============================================
# @File    : snakeCylinder2.py
# @Date    : 2024-07-31 16:06
# @Author  : 帅宇昕
# ============================================
import pybullet as p
import time
import math

# This simple snake logic is from some 15 year old Havok C++ demo
# Thanks to Michael Ewert!
import pybullet_data

p.connect(p.GUI)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
plane = p.createCollisionShape(p.GEOM_PLANE)

p.createMultiBody(0, plane)

useMaximalCoordinates = True
# mouse_force, mouse_torque = p.getMouseForce()
# colBallId = p.createCollisionShapeArray([p.GEOM_BOX, p.GEOM_SPHERE],radii=[sphereRadius+0.03,sphereRadius+0.03], halfExtents=[[sphereRadius,sphereRadius,sphereRadius],[sphereRadius,sphereRadius,sphereRadius]])
# 修改为一个球体

ball_ids=list()

cableLen=5
ballNum = 100
height = 0.02
sphereRadius=0.01
preLinkId=-1
baseAngle=[0,0,0]
basePose=[0.0,0.0,0.0]
colBallId = p.createCollisionShape(p.GEOM_CYLINDER,radius=sphereRadius,height=height)

# 缆线的质量
mass = 0.01
# 可视化的数据id，-1就是默认颜色，不自定义可视化
# visualShapeId = -1
visualShapeId=p.createVisualShape(p.GEOM_CYLINDER,radius=sphereRadius,length=height*1.8,rgbaColor=[1,0,0,1])
endpointsVisualShapeId=p.createVisualShape(p.GEOM_CYLINDER,radius=sphereRadius,length=height,rgbaColor=[1,0,0,1])

basePosition=[0+basePose[0], height+basePose[1] , 1+basePose[2]]
baseOrientation = p.getQuaternionFromEuler(baseAngle)

linkMasses=list()
linkCollisionShapeIndices=list()
linkVisualShapeIndices=list()
linkPositions=list()
linkOrientations=list()
linkInertialFramePositions=list()
linkInertialFrameOrientations=list()
linkParentIndices=list()
linkJointTypes=list()
linkJointAxis=list()

for i in range(ballNum):
    linkMasses.append(mass)
    linkCollisionShapeIndices.append(colBallId)
    if i == 0 or i == ballNum-1:
        linkVisualShapeIndices.append(endpointsVisualShapeId)
    else:
        linkVisualShapeIndices.append(visualShapeId)
    linkPositions.append([0,0,height])
    linkOrientations.append(p.getQuaternionFromEuler([0,0,0]))
    linkInertialFramePositions.append([0,0,0])
    linkInertialFrameOrientations.append(p.getQuaternionFromEuler([0,0,0]))
    linkParentIndices.append(i)
    linkJointTypes.append(p.JOINT_FIXED)
    linkJointAxis.append([0,0,0])

tmp=p.createMultiBody(
    baseMass=mass,
    baseCollisionShapeIndex=colBallId,
    baseVisualShapeIndex=endpointsVisualShapeId,
    basePosition=basePosition,
    baseOrientation=baseOrientation,

    linkMasses=linkMasses,
    linkCollisionShapeIndices=linkCollisionShapeIndices,
    linkVisualShapeIndices=linkVisualShapeIndices,
    linkPositions=linkPositions,
    linkOrientations=linkOrientations,
    linkInertialFramePositions=linkInertialFramePositions,
    linkInertialFrameOrientations=linkInertialFrameOrientations,
    linkParentIndices=linkParentIndices,
    linkJointTypes=linkJointTypes,
    linkJointAxis=linkJointAxis,
    useMaximalCoordinates=useMaximalCoordinates

)
ball_ids.append(tmp)
print(tmp)

p.setGravity(0, 0, -10)
# p.setRealTimeSimulation(0)
#

#

count=400
# 调整仿真参数
p.setPhysicsEngineParameter(fixedTimeStep=0.001, numSolverIterations=800)
p.setTimeStep(1/240)
while True:
    if count==0:
        a = p.loadURDF("husky/husky.urdf", basePosition=[0, 0, 3], baseOrientation=p.getQuaternionFromEuler([0, 0, 0]))
        print(f"husky id:{a}")
        # alen= len(ball_ids)/2
        for ball_id in ball_ids:
            # alen-=1
            # if alen<=0:
            #     break
            p.removeBody(ball_id)

        count-=1

    elif count>0:
        count-=1
    p.stepSimulation()

`

unishuai, 2024-08-07:
i can use it to remove a single body creating by createMultibody

biosbeta, 2024-08-17:
> i can use it to remove a single body creating by createMultibody
Have you solved it?

unishuai, 2024-08-17:
> > i can use it to remove a single body creating by createMultibody
> > Have you solved it?
yes, if there were a single body,i can use the  function "removebody" to remove it.
unfortunately,if the object were created by many collision bodies , i failed .
maybe i use an error way to deal it.What am I supposed to do?