bulletphysics/bullet3 #4651: pybullet.error: GetBasePositionAndOrientation failed.

wagh311, 2024-08-07:
I used pybullet to create a custom environment class named MultiTrackAviary, and I want to initialize this class, as follows:
`env = MultiTrackAviary()  # Continuous action space
env_evaluate = MultiTrackAviary()`
However, the following error occurred：
`Traceback (most recent call last):
  File "F:\Code\gym-pybullet-drones\gym_pybullet_drones\examples\test2.py", line 6, in <module>
    env_evaluate = MultiTrackAviary()
  File "F:\Code\gym-pybullet-drones\gym_pybullet_drones\envs\MultiTrackAviary.py", line 71, in __init__
    super().__init__(drone_model=drone_model,
  File "F:\Code\gym-pybullet-drones\gym_pybullet_drones\envs\BaseRLAviary.py", line 79, in __init__
    super().__init__(drone_model=drone_model,
  File "F:\Code\gym-pybullet-drones\gym_pybullet_drones\envs\BaseAviary.py", line 227, in __init__
    self._updateAndStoreKinematicInformation()
  File "F:\Code\gym-pybullet-drones\gym_pybullet_drones\envs\BaseAviary.py", line 573, in _updateAndStoreKinematicInformation
    self.target_pos[:], _ = p.getBasePositionAndOrientation(self.target_id, physicsClientId=self.CLIENT)
pybullet.error: GetBasePositionAndOrientation failed.`
Can someone help me？

deng-h, 2024-11-01:
do you fix it?

SourishW, 2025-04-06:
Are you using MacOS? I am facing the same issue

Roboticyang, 2026-09-05:
> Are you using MacOS? I am facing the same issue

I recently upgraded to the ARM64 version of Bullet. I have a similar issue. I think it is related to my multi-threading. I can still operate my program, unless it was terminated or crashed unexpectedly. But rerunning it would still work.