bulletphysics/bullet3 #4659: macos sequoia build failure

chenrui333, 2024-09-12:
👋 trying to build 3.25 release for macos sequoia, but ran into the following build failure.

```
  /opt/homebrew/Library/Homebrew/shims/mac/super/clang++  -std=c++11 -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk -dynamiclib -Wl,-headerpad_max_install_names -o libOpenGLWindow.dylib -install_name @rpath/libOpenGLWindow.dylib CMakeFiles/OpenGLWindow.dir/MacOpenGLWindow.o CMakeFiles/OpenGLWindow.dir/__/ThirdPartyLibs/glad/gl.o CMakeFiles/OpenGLWindow.dir/MacOpenGLWindowObjC.o CMakeFiles/OpenGLWindow.dir/EGLOpenGLWindow.o CMakeFiles/OpenGLWindow.dir/GLFWOpenGLWindow.o CMakeFiles/OpenGLWindow.dir/GLInstancingRenderer.o CMakeFiles/OpenGLWindow.dir/GLPrimitiveRenderer.o CMakeFiles/OpenGLWindow.dir/GLRenderToTexture.o CMakeFiles/OpenGLWindow.dir/LoadShader.o CMakeFiles/OpenGLWindow.dir/OpenSans.o CMakeFiles/OpenGLWindow.dir/SimpleCamera.o CMakeFiles/OpenGLWindow.dir/SimpleOpenGL2App.o CMakeFiles/OpenGLWindow.dir/SimpleOpenGL2Renderer.o CMakeFiles/OpenGLWindow.dir/SimpleOpenGL3App.o CMakeFiles/OpenGLWindow.dir/TwFonts.o CMakeFiles/OpenGLWindow.dir/fontstash.o CMakeFiles/OpenGLWindow.dir/opengl_fontstashcallbacks.o CMakeFiles/OpenGLWindow.dir/__/ThirdPartyLibs/stb_image/stb_image_write.o  -Wl,-rpath,/tmp/bullet-20240910-4882-fr3r8c/bullet3-3.25/builddbl/src/Bullet3Common -framework Cocoa ../../src/Bullet3Common/libBullet3Common.3.25.dylib /Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries/libGL.tbd
  Undefined symbols for architecture arm64:
    "_CGLSetParameter", referenced from:
        _Mac_createWindow in MacOpenGLWindowObjC.o
  ld: symbol(s) not found for architecture arm64
  clang++: error: linker command failed with exit code 1 (use -v to see invocation)
  make[2]: *** [examples/OpenGLWindow/libOpenGLWindow.dylib] Error 1
  make[1]: *** [examples/OpenGLWindow/CMakeFiles/OpenGLWindow.dir/all] Error 2
  make[1]: *** Waiting for unfinished jobs....
```

full build log in here, https://github.com/Homebrew/homebrew-core/actions/runs/10798628921/job/29952539530

mosra, 2024-09-20:
I'm running into the same issue (undefined references to CGL APIs), although with another project. And I suspect it's a CMake / Homebrew issue on Mac, because the same CMake code worked correctly before.

What I discovered is that a Homebrew build with CMake 3.30.3 discovers the location of the OpenGL library as `/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries/libGL.tbd`, while another build, outside of Homebrew, using CMake 3.26, finds it as `/Applications/Xcode-13.4.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework`.

So far I don't know whether it's a regression in CMake, or whether this regression happens only when CMake is running from within Homebrew, or whether it's a combination of both. I'm not a Mac user, and I don't know Homebrew internals very well either, so I don't know what the above means. My last data point is from 2023, where the exact same CMake code worked well when building the exact same Homebrew package.

What I spotted was that Homebrew overrides the C++ compiler to something else than what gets found when running CMake directly, which might or might not be related (as CMake then may find `OpenGL.framework` in a different location as a consequence). Another possibly related change is https://gitlab.kitware.com/cmake/cmake/-/commit/6b01a27f901b5eb392955fea322cde44a1b782a3, which went into CMake 3.30.3.

mosra, 2024-09-20:
Just for the record, for the Bullet Homebrew package itself it was "fixed" by passing `-DCMAKE_FIND_FRAMEWORK=FIRST` to CMake (https://github.com/Homebrew/homebrew-core/pull/189186), and the same was what resolved the problem in my case.

Nevertheless, I still think this is a problem specific to Homebrew, and the fix is thus specific to Homebrew as well.