bulletphysics/bullet3 #4686: cannot use GUI in ubuntu 24.04

ninagestalt, 2025-02-11:
It worked for me before but since my upgrade to Ubuntu 24.04 I cannot open the pybullet gui anymore (may not be related though, not sure). I am working locally on my laptop without docker or anything. 

I just want to run this simple example:
```
import pybullet as p
import time
import os 

os.environ["DISPLAY"] = ":0"

p.connect(p.GUI)

time.sleep(6)
p.disconnect()
```
This is my output, but no GUI opens:
```
pybullet build time: Nov 28 2023 23:45:17
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Failed to retrieve a framebuffer config
```

I already tried adding the display env var, upgrading pybullet, running other guis like glxgears or xeyes works. 
Any ideas how to solve this?

VincentWanghh, 2025-04-11:
Same problem! May I kindly ask whether you have solved it?
Thanks :)

chinsengi, 2025-04-13:
Same problem. Also using Ubuntu 24.04

gautamsreekumar, 2025-04-14:
Using these environment variables that I got from [another Github issue](https://github.com/Genesis-Embodied-AI/Genesis/issues/219#issuecomment-2654993183) worked for me.

`__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia`

My system has one integrated graphics card and an NVIDIA GPU.

arslanamjad19, 2025-08-16:
Having the same problem. Also using Ubuntu 24.04. Has anyone solved it for system without NVIDIA GPU?