bulletphysics/bullet3 #4712: Failed building wheel for pybullet

FarnazAdib, 2025-04-29:
I am trying to install pybullet in a virtual environment and it fails. I use VS code on a Mac with Sequoia 15.4.1.

I have the latest version of xcode installed on my Mac  and I have tried all options below
```
pip install pybullet
pip3 install pybullet
pip install --only-binary :all: pybullet
pip install pybullet --use-pep517 --no-build-isolation

```

None of them works and I get this error

>       In file included from In file included from Extras/Serialize/BulletFileLoader/btBulletFile.cppExtras/Serialize/BulletFileLoader/bDNA.cpp::1617:
>       :
>       In file included from In file included from Extras/Serialize/BulletFileLoader/btBulletFile.hExtras/Serialize/BulletFileLoader/bDNA.h::1919:
>       In file included from In file included from :
>       In file included from Extras/Serialize/BulletFileLoader/bChunk.cppExtras/Serialize/BulletFileLoader/bCommon.h::2118:
>       :
>       src/LinearMath/btAlignedObjectArray.hIn file included from :Extras/Serialize/BulletFileLoader/bFile.h38::1910:
>       :In file included from  Extras/Serialize/BulletFileLoader/bCommon.hfatal error: In file included from 'new' file not found
>       Extras/Serialize/BulletFileLoader/bFile.hExtras/Serialize/BulletFileLoader/bFile.cpp::1915:
>       :
>       In file included from In file included from Extras/Serialize/BulletFileLoader/bCommon.hExtras/Serialize/BulletFileLoader/bFile.h::2119:
>       :
>       src/LinearMath/btAlignedObjectArray.hIn file included from :Extras/Serialize/BulletFileLoader/bCommon.h38::2110:
>       :src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found   :21:
>       src/LinearMath/btAlignedObjectArray.h:3838 | :
>       fatal error: 'new' file not found
>       #   38 | include 10<:n efatal error: w'new' file not found>
>            /   38# | i#nc38 | #i/nlculdue <new>  /dfeo ri plac/f<neewm>ent new
>       nocr p      l| l         ^~~~~a
>       cement new
>         /ude       | <         ^~~~~n
>       ew>  //for/for placement new
>             |          ^~~~~
>        placement new
>             |          ^~~~~
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       error: command '/usr/bin/clang++' failed with exit code 1
>       1 error generated.
>       1 error generated.
>       Extras/VHACD/src/VHACD.cpp:18:10: fatal error: 'algorithm' file not found
>          18 | #include <algorithm>
>             |          ^~~~~~~~~~~
>       Extras/VHACD/test/src/main_vhacd.cpp:17:10: fatal error: 'algorithm' file not found
>          17 | #include <algorithm>
>             |          ^~~~~~~~~~~
>       1 error generated.
>       In file included from Extras/VHACD/src/vhacdVolume.cpp:16:
>       In file included from src/LinearMath/btConvexHullComputer.h:19:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not foundIn file included from Extras/Serialize/BulletWorldImporter/btWorldImporter.cpp:16:
>       In file included from Extras/Serialize/BulletWorldImporter/btWorldImporter.h:21:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       In file included from Extras/Serialize/BulletWorldImporter/btBulletWorldImporter.cpp:16:
>       In file included from Extras/Serialize/BulletWorldImporter/btBulletWorldImporter.h:19:
>       In file included from Extras/Serialize/BulletWorldImporter/btWorldImporter.h:21:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       In file included from Extras/VHACD/src/vhacdMesh.cpp:17:
>       In file included from src/LinearMath/btConvexHullComputer.h:19:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       In file included from Extras/Serialize/BulletWorldImporter/btMultiBodyWorldImporter.cpp:1:
>       In file included from Extras/Serialize/BulletWorldImporter/btMultiBodyWorldImporter.h:4:
>       In file included from Extras/Serialize/BulletWorldImporter/btBulletWorldImporter.h:19:
>       In file included from Extras/Serialize/BulletWorldImporter/btWorldImporter.h:21:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       In file included from examples/ExampleBrowser/CollisionShape2TriangleMesh.cpp:2:
>       In file included from examples/ExampleBrowser/CollisionShape2TriangleMesh.h:4:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       1 error generated.
>       In file included from In file included from Extras/VHACD/src/vhacdManifoldMesh.cppExtras/VHACD/src/vhacdICHull.cpp:15:
>       In file included from Extras/VHACD/inc/vhacdICHull.h:18:
>       In file included from Extras/VHACD/inc/vhacdManifoldMesh.h:20:
>       Extras/VHACD/inc/vhacdVector.h:18:10: fatal error: 'iostream' file not found
>          18 | #include <iostream>
>             |          ^~~~~~~~~~
>       :15:
>       In file included from Extras/VHACD/inc/vhacdManifoldMesh.h:20:
>       Extras/VHACD/inc/vhacdVector.h:18:10: fatal error: 'iostream' file not found
>          18 | #include <iostream>
>             |          ^~~~~~~~~~
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       In file included from examples/ExampleBrowser/GL_ShapeDrawer.cpp:22:
>       In file included from examples/ExampleBrowser/GL_ShapeDrawer.h:22:
>       src/LinearMath/btAlignedObjectArray.h:38:10: fatal error: 'new' file not found
>          38 | #include <new>  //for placement new
>             |          ^~~~~
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       1 error generated.
>       [end of output]
>   
>   note: This error originates from a subprocess, and is likely not a problem with pip.
>   ERROR: Failed building wheel for pybullet
>   Running setup.py clean for pybullet
> Failed to build pybullet
> ERROR: Failed to build installable wheels for some pyproject.toml based projects (pybullet)

AnnabellaMacaluso, 2025-05-11:
Same issue on Sequoia 15.3.1

arcymonka, 2025-05-14:
I had the same issue on macOS Sequoia 15.4.1. What worked for me was switching to Conda and installing pybullet from the conda-forge channel, which provides precompiled binaries and avoids the build problems:

```
conda create -n pybullet-env -c conda-forge python=3.11 pybullet
conda activate pybullet-env
```
This solved the problem completely on my M2 Mac. Hope this helps!

Muxi-Qiao, 2025-05-17:
I had the same issue on macOS Sequoia 15.5 with M4 Apple Silicon and I used python==3.10.15. I used conda-forge channel to install it. Although it worked, it also changed my python version to 3.10.13.

AlexisMotet, 2025-05-18:
had a similar problem because the clang in my path was too new, I ran "CC=/usr/bin/clang CXX=/usr/bin/clang++ pip install pybullet" and it worked

gonzalocasas, 2025-06-06:
Same here. I would ideally like to be able to install on the Python interpreter that ships with Rhino (CPython 3.9). Also Mac on arm64, Sequoia 15.3.2

rafaelcaricio, 2025-07-11:
Indeed I can copile and install PyBullet on my M1, but it keeps spinning the mouse. Sounds like a busy loop somewhere.

entangeled, 2025-07-30:
> I had the same issue on macOS Sequoia 15.4.1. What worked for me was switching to Conda and installing pybullet from the conda-forge channel, which provides precompiled binaries and avoids the build [...]

Hi  ! For this - I tried to hack the conda. so that I can use pybullet in in verv. but it did not let me - do you know how I can fix it - or how I can get pybullet in verv. . the code you sent did not work on mine

mbUSC, 2025-09-08:
+1 Same issue here. I don't know what to do to fix this :(